Xinyuan-LilyGO/LilyGo-T-PCIE #1: esp idf pppos

coffice12, 2020-07-15:
esp-idf/examples/protocols/pppos_client

run error:

ts Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6924
load:0x40078000,len:14368
load:0x40080400,len:3980
entry 0x4008067c
I (29) boot: ESP-IDF v4.2-dev-2243-gcf056a7d0-dirty 2nd stage bootloader
I (29) boot: compile time 11:40:59
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1885c (100444) map
I (152) esp_image: segment 1: paddr=0x00028884 vaddr=0x3ffb0000 size=0x02338 (  9016) load
I (156) esp_image: segment 2: paddr=0x0002abc4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/cafe/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (159) esp_image: segment 3: paddr=0x0002afcc vaddr=0x40080400 size=0x0504c ( 20556) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x62b3c (404284) map
0x400d0020: _stext at ??:?

I (329) esp_image: segment 5: paddr=0x00092b64 vaddr=0x4008544c size=0x06628 ( 26152) load
0x4008544c: print_backtrace_entry at /Users/cafe/esp-idf/components/esp_system/port/panic_handler.c:144

I (347) boot: Loaded app from partition at offset 0x10000
I (347) boot: Disabling RNG early entropy source...
I (359) cpu_start: Pro cpu up.
I (359) cpu_start: Starting app cpu, entry point is 0x40081110
0x40081110: call_start_cpu1 at /Users/cafe/esp-idf/components/esp_system/port/cpu_start.c:109

I (345) cpu_start: App cpu up.
I (373) cpu_start: Pro cpu start user code
I (373) cpu_start: Application information:
I (373) cpu_start: Project name:     pppos_client
I (379) cpu_start: App version:      v4.2-dev-2243-gcf056a7d0-dirty
I (386) cpu_start: Compile time:     Jul 15 2020 11:40:57
I (392) cpu_start: ELF file SHA256:  b8479d6d56fbed52...
I (398) cpu_start: ESP-IDF:          v4.2-dev-2243-gcf056a7d0-dirty
I (405) heap_init: Initializing. RAM available for dynamic allocation:
I (412) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (418) heap_init: At 3FFB3CB8 len 0002C348 (176 KiB): DRAM
I (424) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (431) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (437) heap_init: At 4008BA74 len 0001458C (81 KiB): IRAM
I (444) spi_flash: detected chip: generic
I (448) spi_flash: flash io: dio
I (453) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (465) uart: queue free spaces: 30
E (975) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (975) dce_service: esp_modem_dce_sync(48): send command failed
E (975) sim800: sim800_init(460): sync failed
E (1475) esp-modem: esp_modem_dte_send_cmd(229): process command timeout
E (1475) dce_service: esp_modem_dce_sync(48): send command failed
E (1475) sim800: sim800_init(460): sync failed
assertion "dce != NULL" failed: file "/Users/cafe/esp-idf/examples/protocols/pppos_client/main/pppos_client_main.c", line 269, function: app_main

abort() was called at PC 0x400d7777 on core 0
0x400d7777: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x400850b7:0x3ffb5950 0x400857f9:0x3ffb5970 0x400896ae:0x3ffb5990 0x400d7777:0x3ffb5a00 0x400d3272:0x3ffb5a30 0x40087570:0x3ffb5b60 0x400874d5:0x3ffb5b80
0x400850b7: panic_abort at /Users/cafe/esp-idf/components/esp_system/panic.c:349

0x400857f9: esp_system_abort at /Users/cafe/esp-idf/components/esp_system/system_api.c:105

0x400896ae: abort at /Users/cafe/esp-idf/components/newlib/abort.c:46

0x400d7777: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d3272: app_main at /Users/cafe/esp-idf/examples/protocols/pppos_client/main/pppos_client_main.c:269 (discriminator 1)

0x40087570: main_task at /Users/cafe/esp-idf/components/freertos/xtensa/port.c:511

0x400874d5: vPortTaskWrapper at /Users/cafe/esp-idf/components/freertos/xtensa/port.c:169

ELF file SHA256: b8479d6d56fbed52

LilyGO, 2020-07-15:
GitHub示例是Arduino环境的。为什么要用ESP-IDF

LilyGO, 2020-07-15:
https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/get-started/index.html
这里是官方写的ESP-IDE中文文档 可以参考

coffice12, 2020-07-15:
我的项目是esp idf。另外确认一下  pcie的 7600模块的rts  和cto的pin是多少？还有我也有你们公司的sim800模块
https://detail.1688.com/offer/617438947038.html?spm=a26352.13672862.offerlist.1.6ec36bf3WhgzjV
这个 他的的rts  和cto的pin是多少？

LilyGO, 2020-07-15:
TX和RX    README的引脚图写了。请注意查看。
ESP-IDF问题查阅官方文档

coffice12, 2020-07-15:
我这里问的不会tx和rx是问 rts和cts两个pin的

closed 2020-07-22 by coffice12