Xinyuan-LilyGO/LilyGo-T-PCIE #7: Sim module stability

pm4r, 2020-11-03:
I have board with Sim7070G and have a problem with sim module stability. I suspect current is too low. I'm able to power up and connect to sim module through uart. AT commands works well, sim card is recognized, I can check firmware, etc. but module can't register to network and restarts periodically. In fact I can only issue AT commands for a couple of seconds before it spits '\x00' characters and restarts.
I doesn't matter if I power it from USB, Li-Ion or both. I have checked current drawn from battery (through APX192 and also with external ammeter) and it's below 100mA with peaks up to 160mA which is way to low for sim module to work properly. ESP32 works fine and never browns out.

As I understand from schematics (https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/schematic/T_PCIE_Board.pdf) sim module is powered from VSYS (AXP IPSOUT pins) and not through any DCDC, so how can I adjust current? Do you have any specific AXP192 configuration - I can't find datasheet in English.

I'm using latest Micropython firmware.

whitesmokepcb, 2021-06-30:
Unfortunately looks that board have some serious issue with current, we remove AXP192 chip at all and add additional super-caps to make SIM868 and 7070 work stable.  Lost lot of the time with lab power supply and oscilloscope to run this...

LilyGO, 2021-10-21:
Hello, the board looks normal here, can you describe the process of reproducing this issue?

star297, 2021-11-24:
I have SIM7070G modules, three of them (not PCI-e versions) and cant get them to work in the EU or at least in North Spain.
Perhaps no NB-IOT where we are?
Same issue, I can do everything except connect to network.
The SIM7070G is a waste of time anyway, you can't use GSM and GNSS at the same time due to the device shares the RF internal circuits so you have to switch off GSM part to get GPS information. 
Great is you only want very occasional GPS position data, no good for real time tracking.
We use SIM7600 PCI-e and SIM7000 PCI-e modules, both work well with these LillyGO boards.
SIM7600 has a better GNSS sensitivity and quicker to acquire a 3d fix.
We have had no issue with power however we use a carrier PCB that has added solid aluminium capacitors (47uf/6.3v) on the 5v-IN and 3.3v pins.

PyromanV, 2022-01-07:
I know that this is a very old issue, but I would rather not open a new one to post a potential solution. I've recently got one of those boards and probably experienced the same problem, however, I managed to fix it with a small hardware mod (which is not ideal, but hey, as long as it works and it isn't anything as drastic as removing the AXP192), so this is more of an tips n tricks post for others potentially debugging the same issue (this is not a unique problem as far as I've seen in T-PCIe's closed issues).

My T-PCIe is running on battery power with SIM868, and initially I couldn't get it to join the network while powered from battery (SIM module brownout, battery voltage 3.7V). So after trying multiple startup delays, adding caps in multiple places and other stuff, I settled on adding a 470u polarized cap in parallel with another polarized one found on the modem daughter-board (EL5 in schematic - the one next to a group of small caps and a diode). After this mod, the whole board wouldn't even start when running example sketch (probably AXP192 overcurrent) - but this was easily fixed by adding a couple of seconds delay right after ESP32 startup and another couple seconds before powering up the regulator (pin 25). I have yet to test this setup extensively.

Michel-Mayoud, 2022-12-14:
From all these comments, I make the connection beetween this issue to an old one that I had with the LilyGo T-call Sim800 board. The circuit control power isn't able to supplie the current requiered by the SIM controler and the communication doesn't work correctly. I had some hope discovering this new shied with Sim module compatible with 4G, doesn't work better than the old generation. I'm really disapointed by Lilygo which can't take advatange from issue and goes to produce defective products.