Xinyuan-LilyGO/LilyGo-T-PCIE #13: Wrong schematic! Please update it!

Nefarious19, 2021-05-14:
Schematic is outdated. For example there is no modem  wwan led on the board or it's not connected. Please update schematic!

randybb, 2021-05-28:
People are asking for updated schematics and they are just ignoring it.

Anyway, this whole board is so badly designed. They haven't used the whole potential of axp192 for powering modem or other peripherals and there is no way to tap the other not connected DCDC/LDO (at least they could connect these to test points). As they are selling the modem to end users they have to provide firmware updates - never responded to me. Totally waste of money. RAK WisBlock or the newest Particle B series (B524) / Tracker One (ONE524MEA) is much better option...

LilyGO, 2021-09-25:
The schematic has been updated, if you have any further questions, feel free to contact

closed 2021-09-25 by LilyGO