Xinyuan-LilyGO/LilyGo-T-PCIE #14: I2C connection to modem

FlorianBogeschdorfer, 2021-05-17:
Unfortunately the documentation is not existing for the I2C on the modem. I have a SIM7600E modem PCIE card by Lilygo - coulnd't find anything on that. However I measured and fund that the I2C SCL and SDA from pins 55 and 56 of the modem are connected to the PCIE pins 30 and 32. On the Wrover board these are even named SDA and SCL (printed on the board). However these pins seem to be not connected. In the documentation ist says TP7 and TP8. What are TP7 and TP8? I measured - there is no connection to the I2C from the wrover (21,22). And apparently not to any other pin.

LilyGO, 2021-05-28:
There is a schematic diagram in the README

LilyGO, 2021-05-28:
There are pin diagrams in the README

LilyGO, 2021-09-25:
The I2C interface of the module does not lead to the motherboard.

closed 2021-09-25 by LilyGO