Xinyuan-LilyGO/LilyGo-T-PCIE #15: Unable to use CAN

zeljkoavramovic, 2021-06-09:
It seams that ESP has fixed ping GPIO4 and GPIO5 for CANH and CANL. I wanted to connect them to SN65HVD230 for logging car OBDII CAN messages, but it seams that GPIO is mapped to M.2 which I wanted to use for 4G.

If my understanding about GPIO5 is correct, could you please fix this in some future revision?

romoloman, 2021-08-14:
GPIO5 is free is GPIO4 that is in use for powerkey (for powering on the PCI modem).
I'm using GPIO5 in one of my project and it works perfectly.
You can cut the track from GPIO4 to powerkey and connect to any other free pin if you need it...

zeljkoavramovic, 2021-08-14:
Thanks, but I've already switched to custom PCB and gave up on LilyGO.

closed 2021-08-14 by zeljkoavramovic