Xinyuan-LilyGO/LilyGo-T-PCIE #16: Missing audio decoder chip

t0bybr, 2021-06-30:
Hi,
I'm thinking about making a basic mobile phone with integrated GPS tracker. Your boards seem to be great for this. Especially the LilyGo T-PCIe. But before ordering I have two questions.

[On this site](https://www.aliexpress.com/item/4001142716386.html?spm=a2g0o.store_pc_groupList.8148356.11.63e736fcVYSjXm) this is stated: *The product does not contain audio decoder chip NAU8810YG* Does that mean it is not possible to make voice calls with this modem?
And I see a Motor connector, is this for a vibration motor?

Thanks in advance.
t0bybr

LilyGO, 2021-09-25:
Sim868 can make phone calls without decoding chip

LilyGO, 2021-09-25:
Sim7600 needs decoding chip, there is no default paste, you can ask customer service to add this chip

closed 2021-10-21 by LilyGO