Xinyuan-LilyGO/LilyGo-T-PCIE #17: reading battery voltage

tradexsrl, 2021-08-06:
Hi is it possible to read battery voltage as in lilygo ttgo t-sim7600E ?
any sample code ?

romoloman, 2021-08-13:
int16_t vscale=1000;
int16_t vbatt=(int16_t)((modem.getBattVoltage()*vscale)/1000);
where vbatt is voltage in millivolt and vscale is a calibration factor (start with 1000 and adjust);

Please keep in mind that when charging you will always read 4000 + as value, also when connected with usb.
To have a good reading you must run on battery and display the value using a webserver or something like that or use a serial port without using the USB on board.

When you are running on battery measure the battery voltage with a multimeter and adjust vscale,

tradexsrl, 2021-08-13:
Many thanks, I supposed you have tested the board and one pci-e modem like an nb-iot or an lte standard or you had it in your hands , have you found any issue or not? Can you recommend this board instead of lilygo ttgo t-sim 7600e for proof of concept for an industrial and smart agriculture device?

romoloman, 2021-08-14:
I'm using this board with a sim7600e on an industrial device without any single issue since months....

tradexsrl, 2021-08-14:
Perfect , many thanks

closed 2021-08-14 by tradexsrl