Xinyuan-LilyGO/LilyGo-T-PCIE #21: Use with LoRa concentrator?

serbyxp, 2022-01-30:
I have LoRa sx1302 concentrator from seeed studio. It has USB option or SPI option, I bought the T-PCIe module, and received already. I am wondering more information about the Pin out for the Mini-PCIe slot. If I can use with the USB version of LoRa concentrator it uses STM32 for USB on pins 36 USB - NS pin 38 USB+ 
I can also use SPI pins 45 spi_clk 47 spi_miso 49 spi_mosi 51 chip select I am not using the sx1262 radio only sx1250 radios which uses pin 22 for Reset  called 1302_reset in diagram bellow I will also need to power and ground pins. I would like to use USB but if not I can use SPI. Is this possible with the pins for mPCIe ? I see pinout for ESP32 on diagrams but not the pins for mPCIe.
Thank you.
[image]

serbyxp, 2022-02-07:
The reason I ask is because I see this in the Schematic, But I am not clear on  the parts I put a Star Or Arrow...
[image]
[image]
[image]
[image]

Kind of unsure of this, let me see if I can explain or if anyone knows if they can clarify this for me.

ESP32 diagram
U0TX / RX are on ESP32 pins 35 / 34     They are referenced to the UART diagram
U0TX / RX are on UART  pins 21/20       Which has a DM / DP on D- / D+ on the UART diagram then on the USB diagram
DM / DP are on a led resistor circuit which starts with a D+ / D- Which then goes back to the actual USB connector which
D- / D+ which I am understanding with USB type C socket has a top and bottom or A and B do both A (DP1 DN1)- / D+ and B D- / D+  (DN DP), connect together? I can kind of see on the actual PCB That there are Traces / Maybe through the VIAs on another layer connect?
IF anyone knows which D- / D+ (A / B) is going to the DM DP or is it both? because the DM DP ends at D- D+ . Open VBUS solder points on the other side of the PCB,  on the schematic for MPCIE diagram shows something named USB_DP / USB_DM connects to VBUS pin 36 and 35. Which are the pins that I need for my concentrator card to work on USB so I can use the STM32 chip thats on my concentrator, as not to use the ESP32 CPU for those functions, Just to pass the UDP packets over WIFI.

That would be super helpful before I start coding all this and get stuck, Or maybe I just jump some cables over back into the USB-C from the Dbus I dont know that seems complicated, IF anyone can help that would be appreciated Thank you.

LilyGO, 2022-02-09:
Hello, ESP32-Wrover has no USB port. SPI is recommended

serbyxp, 2022-02-09:
I see the usb to uart chip do you mean the mPCIe pins that are usb_DM DP , don’t connect to the uart chip that connects to the esp32 uart? If I plug USB_DM DP from vbus unpopulated pins into usb type c pins, it will not work through the cp2140 chip ?

LilyGO, 2022-02-10:
As shown in figure.  DW, DP are derived from modules (e.g. Sim7000, SIM7600.....)  .  Not esp32 pins  
[image]

serbyxp, 2022-02-10:
Yes, my Module ( concentrator  sx1302) has STM32 on the same MPCIe pins as SIMxx. My Question was does the USB_DM USB_DP in the diagram above end at the p1 header? Or does it connect through VIAs or Traces anywhere else on the board? 

USB_DM and USB_DP are not the same as the DM and DP pins from the cp2104, correct? I understand I can use the SPI I will try that. I wanted to try and use the STM32 on the LoRa module with ESP32 for DEV purpose, I will try with adding the Jumper cables to custom USB-C connector to jumper USB_DM USB_DP -> USB-C -> CP2104 -> ESP32 -> 2.4ghz , as well as the SPI -> ESP32- 2.4ghz.

Thank you.

LilyGO, 2022-02-11:
Yes, USB_DM and USB_DP are different from CP2104's DM and DP pins

mailinger-mate, 2022-05-24:
Hi @serbyxp! Have you been able to learn more about using the SX1302 concentrator with the T-PCIE?

serbyxp, 2022-06-07:
> Hi @serbyxp! Have you been able to learn more about using the SX1302 concentrator with the T-PCIE?

only thing I found was micropython sx1308 sketch that might be able to be modified with sx1302 registers and src files etc … but my concentrator is a usb version not spi version so their is some things I would need to work out with the above issues of the usb dm do pins to re inject them into the cp ch serial , or change the code in the stm32 to not be a virtual com port. It’s kind of on the back burner at the moment. The spi version might be easier to use and port the spi to WiFi espressif library to act / replace the com libraries on the lora-net sx1302 hal… but I haven’t re dived into it . Another option I saw was possibly to use the micropython make library to compile to lora-net sx1302 hal into a micropython module . But I didn’t get that far yet. Let me know if you found anything. The micropython library with sx1308 to compare the build files / make files and src inc files is the the pycom library sigfox lorawan micropython main library I don’t want to get in trouble posting that here … but it’s in there . Main thing I had issues is that esp32 uses cmake instead of MakeFile so o got tripped up there .