Xinyuan-LilyGO/LilyGo-T-PCIE #22: t-pcie with sim7600 gps get latitude and longitude problem

tradexsrl, 2022-02-16:
Hi , i was trying T-PCIE/examples/SIM7600/TinyGSM_Allfunctions/ examples with latest arduino ide and latest tinygsm library but i never get lat and lon.....
i setup gnss mode to galileo because i found by default gps module is configured for non european gps system but even let it run for 30 minutes the sketch i never get a signal lock from satellite
please let me know

LilyGO, 2022-02-18:
Hello，What does the serial port print? Make sure the antenna is outdoors

tradexsrl, 2022-02-18:
Hi , i was able to get lat lon after many hours of running the example but the result are negative..... for example lat -45 , lon -9

LilyGO, 2022-02-21:
Are you not changing TinyGSM_Allfunctions?  Would you like to try this firmware? 
[PCIE_SIM7600X.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/files/8108044/PCIE_SIM7600X.zip)

tradexsrl, 2022-02-21:
hi , no changes on example tinygsm_all functions , i have sim7600E module , this firmware is for this one? because i see 7600X

LilyGO, 2022-02-23:
7600E can use the firmware sent above.  Sim7600 cannot be upgraded in other modes.

tradexsrl, 2022-02-23:
ok for the file , can i use this adapter to do the upgrade ?
https://techship.com/products/mpcie-to-usb-31-adapter/

tradexsrl, 2022-02-23:
in the mean time can you test the 7600 simcom module to see with the example provided if you get lat and lon negative ?
i don't want the problem could be the tinygsm library

LilyGO, 2022-02-23:
What should your GPS data be for the correct location?  The upgrade is just a matter of pulling out the USB cable

tradexsrl, 2022-02-23:
correct latitude is 45 , correct longitude is 9 
i get latitude -45 , longitude -9

LilyGO, 2022-02-24:
Use what firmware I sent you. Is the result the same?

tradexsrl, 2022-02-24:
What version of firmware have you sent me? Because i have no way to do the upgrade but I'm able to get via at command the firmware I already had in the simcom module....maybe they are the same or maybe your fw is older than what I have in the simcom module ...and in this case the problem of getting negative lat Lon is still there

tradexsrl, 2022-02-25:
this is the actual firmware: **SIMCOM_SIM7600E Revision: SIM7600M21-A_V2.0.1**
if the firmware you sent me is the same version , then is not usefull to try the upgrade and the problem still there

LilyGO, 2022-02-26:
https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/files/8108044/PCIE_SIM7600X.zip
This is for burning firmware for esp3
[image]
2

tradexsrl, 2022-02-28:
tried with no luck...i have connected the t-pcie to my pc via usb cable with simcom7600 module mounted on top with simcard inserted with no pin, then i run the downloader 3.8.8 , setup com4 as the one recognized by my pc , then i setup all the things as in your screenshot but i get this one only :
[image]

tradexsrl, 2022-02-28:
and never get to finish in green as in your screenshot

LilyGO, 2022-03-01:
Click "START" no normal burning? What about reopening the tool?

tradexsrl, 2022-03-01:
i click on start....and i get stop in brown....i tried again now , same problem.
i suspect the tools needs a way to go through esp32 and get to simcom7600 module via a softwareserial...so you need a sketch to load to esp32 and enable this passthrough

LilyGO, 2022-03-02:
This is the firmware for the ESP32. Connect type-c directly

tradexsrl, 2022-03-02:
i don't understand...the connector type-c on the t-pcie board is connected to my pc...there is no any other connector on the board...

LilyGO, 2022-03-03:
No ports?  What kind of computer do you have, with a drive?  Does the board start normally?

tradexsrl, 2022-03-03:
hi , i think we don't understand each other....simcom7600 is mounted on t-pcie board and this is connected to my pc but with download tool i have the problem written...see the picture
[image]

LilyGO, 2022-03-04:
You are right to connect. Is there a port? Can it burn normally?

tradexsrl, 2022-03-04:
yes in my pc there is com4 and i can program normally with arduino IDE but NOT with the tool suggested by you to upgrade the simcom firmware.
let me know once and for all with the correct and detailed procedure to do that.