Xinyuan-LilyGO/LilyGo-T-PCIE #23: Powering from 5V pin whilst debugging via USB

ntsewell, 2022-04-25:
Hi!
We are developing a product around the LilyGo-T-PCIE board and wish to confirm whether it is possible to power board from the 5V pin whilst at the same time having the USB-C connected to a computer for debugging / monitoring.  Given the dev nature of this board it would seem sensible to be able to do this, but we want to confirm.

Thank you!

randybb, 2022-04-25:
Based on the last page of schematic https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/schematic/T_pcie_v1.1.pdf there are no protection diodes, so no.

ntsewell, 2022-04-25:
That's what we thought too... shame, it would be a useful and very cost effective addition!  Thanks Randy.

carlosprato1, 2022-05-29:
As @randybb says, it does not have protection diodes, maybe it works to place them on the usb cable and on the power supply. something like that: [sources_diodes](http://blog.fullwat.com/wp-content/uploads/2020/05/fuentes-paralelizar-diodos-300x207.jpg)  .

I got the image from [this page](http://blog.fullwat.com/faq-fuentes-de-alimentacion-parte-3/#:~:text=El%20circuito%20en%20paralelo%20es,el%20voltaje%20se%20mantiene%20igual.): 

maybe one day I'll try it in the meantime I'll keep disconnecting the source to connect the USB on the board.

mereshow, 2022-06-18:
> Hi! We are developing a product around the LilyGo-T-PCIE board and wish to confirm whether it is possible to power board from the 5V pin whilst at the same time having the USB-C connected to a [...]
I do it often with no problem. I use a laptop connected to the same electrical circuit as the esp32 power source, or use the laptop unplugged (battery only).

ntsewell, 2024-01-08:
> I do it often with no problem. I use a laptop connected to the same electrical circuit as the esp32 power source, or use the laptop unplugged (battery only).

I meant to say thank you for your comment about this - we also do this all the time now so it is safe it seems! :)