Xinyuan-LilyGO/LilyGo-T-PCIE #24: How to customize button1 connected to axp192

tradexsrl, 2022-04-28:
Hi , i need to customize button 1:
[image]

for example keep it pressed for more than 3 seconds , start a softap
can you point me to documentation on how to define it in my sketch ?

LilyGO, 2022-04-29:
Hello.  Button 1 cannot be customized.  Where did you see this picture?   The board also has many IO pins that can be extended.

tradexsrl, 2022-04-29:
i'm your customer for an expansion board you are custom design for us and i'm in contact with sales@lilygo.cc asking if there is a button configurable via software to keep it pressed for example for 3 sec at powerup to start a function for softap.
i get this replay:
It can be defined by program .
[ https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE)
or https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library for reference , about Button .

i was also asking if there are free touch pin for my us case but seems no more free pins because the custom board has 2 terminal for two button + hx711 + dht11 and we are using simcom7600 modem 
let me know

LilyGO, 2022-04-29:
The PWR button is connected to the PWRON pin of the AXP192.  Hold down to wake up or sleep.  This is one of the features of AXP192.  No software configuration required

LilyGO, 2022-05-03:
Software can interrupt detection by pressing "isPEKShortPressIRQ".  [reference ](https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_base/axp_base.ino#L175)

tradexsrl, 2022-05-03:
i tried to load the example but i get this error:
[image]

tradexsrl, 2022-05-03:
board is connected to pc via usb cable and then to arduino ide on com6
i have connected a battery with jst connector provided with T-PCIE board to the relative connector , before connecting usb to pc i see the board powerup with green led on....battery is a 3.7 volt 18650 type , if i long press button 1 it power off the board , if i press again button 1 it powerup the board but i get again that error.please let me know

tradexsrl, 2022-05-03:
if i run an i2c scanner on the same board i get 1 address found on 0x34.
looking into axp20x.h of the library , i see the declaration AXP192_SLAVE_ADDRESS have the correct address...let me know why i get fail error

LilyGO, 2022-05-04:
Try this bin file
[AXP192_base.zip](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/files/8617481/AXP192_base.zip)
。

tradexsrl, 2022-05-04:
i'm using arduino ide...how to load this bin?

tradexsrl, 2022-05-04:
i'm using the axp_base example from https://github.com/lewisxhe/AXP202X_Library/tree/master/examples/axp_base

LilyGO, 2022-05-04:
1. Use the esp32 [tool](https://www.espressif.com/sites/default/files/tools/flash_download_tool_v3.8.5.zip) .  [Refer ](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/firmware#readme)  to burning configuration
2. I compiled using an Arduino IDE.  There are no problems like those in your screenshots

tradexsrl, 2022-05-04:
ok i loaded the bin sent and i get this:
[image]

if i press short or long the button1 , once or twice i get only the reply message isPEKShortPressIRQ....i never get the other message...
my hardware setup is this: 
[image]

now i need to understand why i get error on my example compiled..otherwise i cannot do what i need.
ide is arduino 1.8.19 on windows 10 pro v1909 
board selected is esp32 dev with these setup:
[image]
there is only one axp library and was installed by downloading the zip file from here: https://github.com/lewisxhe/AXP202X_Library
and the sample i tried is the axp_base.ino
please let me know what you have different than me.

tradexsrl, 2022-05-04:
do i need to change these value in the axp base?
const uint8_t i2c_sda = 21;
const uint8_t i2c_scl = 22;
const uint8_t axp_irq_pin = 35;
are you sure these pin number are ok for t-pcie ?

LilyGO, 2022-05-05:
Try this:
/***********Do not use the following code***************/
//const uint8_t i2c_sda = 21;
//const uint8_t i2c_scl = 22;
//const uint8_t axp_irq_pin = 35;

/************Add the following code**********/
#define  i2c_sda  21
#define  i2c_scl    22
#define  axp_irq_pin 35

tradexsrl, 2022-05-05:
ok now it works , very strange...
since i haven't found any doc or explanation on this power control chip , what does mean these output value:
getVWarningLevel1:3449 mV 
getVWarningLevel2:3399 mV 
getPowerDonwVoltage:2600 mV 
getPowerDonwVoltage:2600 mV 
DC2:1800 mv  DC3:3300 mv  LDO2:3300 mv  LDO3:3300 mv  LDO4:DISABLE  Exten:ENABLE

and for t-pcie if i refer to the flash tool page you provide , do i need to setup flash frequency 40Mhz ? and flash mode DIO ?
because the default are 80Mhz and QIO

LilyGO, 2022-05-05:
QIO or DIO depends on the mode selected during compilation.  You should now compile and run properly.  According to your side of the configuration can be.

tradexsrl, 2022-05-05:
since i haven't found any doc or explanation on this power control chip , what does mean these output value:
getVWarningLevel1:3449 mV
getVWarningLevel2:3399 mV
getPowerDonwVoltage:2600 mV
getPowerDonwVoltage:2600 mV
DC2:1800 mv DC3:3300 mv LDO2:3300 mv LDO3:3300 mv LDO4:DISABLE Exten:ENABLE

LilyGO, 2022-05-06:
As it literally means,
getVWarningLevel : When the battery is lower than 3449 mV, there will be an interruption warning (low battery warning).
getPowerDonwVoltage:  When the battery voltage is lower than 2600mv, the device shuts down.
It can be customized with any of the above values.  
Here is a [manual](https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/197) in English with all the answers you need

tradexsrl, 2022-05-06:
many thank's