Xinyuan-LilyGO/LilyGo-T-PCIE #25: stability on Li-Ion battery

alextroto, 2022-07-01:
We're using a Li-Ion rechargeable battery to power T-PCIE + SIM7600E modem. It's working fine as long as the battery it's over 3.7-3.8 V, but under, the board shutdown. Unfortunately that means I can use only at best 50% of the battery capacity. We've tried with different batteries: Panasonic NCR18650B-3300mA, Samsung INR18650-35E-3500mA, same result.
I'm suspecting it's a problem with the modem onboard capacitors, but no idea how to improve the situation. Any help will be great.
Alex

tyeth, 2022-09-28:
I have no real advice myself to offer, but there was a similar situation where the GSM portion of the 7000G causes a 0.6v drop and the solution to reduce that was to use LTE. The hardware "fixes" don't really change the situation for you capacity-wise: https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/147

fgnievinski, 2023-07-07:
it's a known problem, some report using a supercapacitor helps:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/227#issuecomment-1508438887
also make sure the battery cables are "as short as possible, and as thick as possible."