Xinyuan-LilyGO/LilyGo-T-PCIE #26: waitForNetwork fails

github0013, 2022-08-19:
I bought a set of T-PCIE 4MB 9102 and T-PCIE-SIM7600JC -H from
https://www.aliexpress.com/item/4001142716386.html?channel=twinner&gatewayAdapt=glo2jpn

I seems my board is recognizing SIM7600 module, but network registration is not working despite the fact the LTE signals are doing fine.
I tried `modem.setNetworkMode(2)` also, but basically the same results.

#### Can this be possibly my board problem?  or maybe the SIM card I am using?  (I am using this SIM card on my iPhone and working just fine though ...) 

### logs
https://gist.github.com/github0013/a4980521f792212a3f58c77adea8514b

> +CPIN: READY

https://i.gyazo.com/b3c9067c44509f40beda053702189e5c.jpg
READY - ME is not pending for any password

> AT+CNMP=38

https://i.gyazo.com/697ebc82a89942d1fc94a39137fc6441.jpg
38 - LTEOnly

> +CSQ: 22,99

https://i.gyazo.com/61788f1d3d5e0e24f11644d6b7d66320.jpg
https://m2msupport.net/m2msupport/atcsq-signal-quality/
20 something should be "Excellent" signal level.

> +CGREG: 0,2

https://i.gyazo.com/575abacc1cf0a1ae489641e5d5051b38.jpg

0 - disable network registration unsolicited result code
2 - not registered, but ME is currently trying to attach or searching an operator to register to

I sometimes get `+CGREG: 0,0`, but either way, it doesn't seem the registration is working.

### code I used

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_port = /dev/cu.wchusbserial537D0225881
monitor_speed = 115200
build_flags = -DCORE_DEBUG_LEVEL=5
lib_deps = 
	vshymanskyy/TinyGSM@^0.11.5
	vshymanskyy/StreamDebugger@^1.0.1
```

```cpp
#define SerialMon Serial
#define SerialAT Serial1

static const char *TAG = "SIM LTE";
const char apn[] = "YOUR APN";
const char gprsUser[] = "YOUR USER";
const char gprsPass[] = "YOUR PASS";

#include <Ticker.h>
Ticker tick;

#define uS_TO_S_FACTOR 1000000ULL
#define TIME_TO_SLEEP 60

#define PIN_TX 27
#define PIN_RX 26
#define UART_BAUD 115200
#define PWR_PIN 4
#define LED_PIN 12
#define POWER_PIN 25
#define IND_PIN 36

#define TINY_GSM_MODEM_SIM7600
#include <TinyGsmClient.h>

#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);

void setup() {
  SerialMon.begin(115200);
  delay(6000);
  ESP_LOGD(TAG, "starting");

  // POWER_PIN : This pin controls the power supply of the SIM7600
  pinMode(POWER_PIN, OUTPUT);
  digitalWrite(POWER_PIN, HIGH);

  // PWR_PIN ： This Pin is the PWR-KEY of the SIM7600
  // The time of active low level impulse of PWRKEY pin to power on module ,
  // type 500 ms
  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(500);
  digitalWrite(PWR_PIN, LOW);

  // Onboard LED light, it can be used freely
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, LOW);
  // IND_PIN: It is connected to the SIM7600 status Pin,
  // through which you can know whether the module starts normally.
  pinMode(IND_PIN, INPUT);
  attachInterrupt(
      IND_PIN,
      []() {
        detachInterrupt(IND_PIN);
        tick.attach_ms(1000,
                       []() { digitalWrite(LED_PIN, !digitalRead(LED_PIN)); });
      },
      CHANGE);

  ESP_LOGD(TAG, "preparing");
  delay(3000);
  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);

  ESP_LOGD(TAG, "initializing");
  if (!modem.init()) {
    ESP_LOGD(TAG, "init FAILED");
    return;
  }

  ESP_LOGD(TAG, "Model Name: %s", modem.getModemName().c_str());
  ESP_LOGD(TAG, "Model Info: %s", modem.getModemInfo().c_str());
  ESP_LOGD(TAG, "SIM Status: %d", modem.getSimStatus());

  ESP_LOGD(TAG, "setting network mode");
  do {
    delay(500);
  } while (!modem.setNetworkMode(38));

  ESP_LOGD(TAG, "waiting for network");
  if (!modem.waitForNetwork(60000L, true)) {
    ESP_LOGD(TAG, "waitForNetwork FAILED");
    return;
  }

  if (modem.isNetworkConnected()) {
    ESP_LOGD(TAG, "network connected");
  } else {
    ESP_LOGD(TAG, "network NOT connected");
  }

  ESP_LOGD(TAG, "connecting to: %s", apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    delay(10000);
    return;
  }

  bool res = modem.isGprsConnected();
  ESP_LOGD(TAG, "GPRS status: %s", res);

  IPAddress local = modem.localIP();
  ESP_LOGD(TAG, "Local IP: %s", local);

  int csq = modem.getSignalQuality();
  ESP_LOGD(TAG, "Signal quality: %s", csq);

  modem.poweroff();
  ESP_LOGD(TAG, "power off");
}

void loop() {
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  delay(200);
  esp_deep_sleep_start();
}
```

fabiopolar, 2023-09-12:
Hi, try this:

` bool isConnected = false;
  for (int i = 0; i <= 4; i++) {

    uint8_t network[] = {
        2,  //Automatic/
        13, //GSM only/
        38, //LTE only/
        51  //GSM and LTE only/
    };
    Serial.printf("Try %d method\n", network[i]);
    modem.setNetworkMode(network[i]);
    delay(3000);
    
    int tryCount = 60;
    while (tryCount--) {
      int16_t signal =  modem.getSignalQuality();
      Serial.print("Signal: ");
      Serial.print(signal);
      Serial.print(" ");
      Serial.print("isNetworkConnected: ");
      isConnected = modem.isNetworkConnected();
      Serial.println( isConnected ? "CONNECT" : "NO CONNECT");
      if (isConnected) {
        break;
      }
      delay(1000);
      digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    }
    if (isConnected) {
        break;
    }
  }`

fabiopolar, 2023-09-12:
` bool isConnected = false;
  for (int i = 0; i <= 4; i++) {

    uint8_t network[] = {
        2,  //Automatic/
        13, //GSM only/
        38, //LTE only/
        51  //GSM and LTE only/
    };
    Serial.printf("Try %d method\n", network[i]);
    modem.setNetworkMode(network[i]);
    delay(3000);
    
    int tryCount = 60;
    while (tryCount--) {
      int16_t signal =  modem.getSignalQuality();
      Serial.print("Signal: ");
      Serial.print(signal);
      Serial.print(" ");
      Serial.print("isNetworkConnected: ");
      isConnected = modem.isNetworkConnected();
      Serial.println( isConnected ? "CONNECT" : "NO CONNECT");
      if (isConnected) {
        break;
      }
      delay(1000);
      digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    }
    if (isConnected) {
        break;
    }
  }`

fabiopolar, 2023-09-12:
`bool isConnected = false;
for (int i = 0; i <= 4; i++) {

uint8_t network[] = {
	2,  //Automatic/
	13, //GSM only/
	38, //LTE only/
	51  //GSM and LTE only/
};
Serial.printf("Try %d method\n", network[i]);
modem.setNetworkMode(network[i]);
delay(3000);

int tryCount = 60;
while (tryCount--) {
  int16_t signal =  modem.getSignalQuality();
  Serial.print("Signal: ");
  Serial.print(signal);
  Serial.print(" ");
  Serial.print("isNetworkConnected: ");
  isConnected = modem.isNetworkConnected();
  Serial.println( isConnected ? "CONNECT" : "NO CONNECT");
  if (isConnected) {
	break;
  }
  delay(1000);
  digitalWrite(LED_PIN, !digitalRead(LED_PIN));
}
if (isConnected) {
	break;
}
}`

github0013, 2023-11-30:
@fabiopolar so you are saying don't use`waitForNetwork`, but just manually check `isNetworkConnected` periodically?

kyotee89, 2024-02-28:
@github0013 any luck figuring this out. I am working on a sim7600 project where i get the board to power up and send a GET request every 2 hrs but it randomly stops powering on after the first hours. Any guidance appreciated

github0013, 2024-03-02:
@kyotee89 

It's been for a long time, so I may not remember perfectly, but here's what I did, and it worked for me.

```cpp
#define SerialMon Serial
#define SerialAT Serial1

static const char *TAG = "SIM LTE";
const char *apn = "YOUR APN";
const char *gprsUser = "YOUR USER";
const char *gprsPass = "YOUR PASS";

#define uS_TO_S_FACTOR 1000000ULL
#define TIME_TO_SLEEP 60

#define PIN_TX 27
#define PIN_RX 26
#define UART_BAUD 115200
#define PWR_PIN 4
#define LED_PIN 12
#define POWER_PIN 25
#define IND_PIN 36

// https://github.com/vshymanskyy/TinyGSM#diagnostics-sketch
#define TINY_GSM_DEBUG SerialMon
// SIM7600
#define TINY_GSM_MODEM_SIM7600
#include <TinyGsmClient.h>

// debugging
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);

void setup() {
  // for logging
  SerialMon.begin(115200);

  pinMode(POWER_PIN, OUTPUT);
  digitalWrite(POWER_PIN, HIGH);
  delay(1000);

  // SIM7600_Series_PCIE_Hardware_Design_V1.03.pdf
  // 3.2 PERST#
  // The active low level time impulse on PERST# pin to reset module
  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(500);
  digitalWrite(PWR_PIN, LOW);

  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);
  // wait, otherwise AT commands will be ignored
  delay(5000);

  if (!modem.init()) {
    ESP_LOGD(TAG, "初期化失敗");
    return;
  }

  // network, not sure what I am doing...
  modem.sendAT(GF("+CREG=1"));
  ESP_LOGD(TAG, "+CREG=1 %d", modem.waitResponse());

  // GPRS network
  modem.sendAT(GF("+CGREG=1"));
  ESP_LOGD(TAG, "+CGREG=1 %d", modem.waitResponse());

  // LTE network
  modem.sendAT(GF("+CEREG=1"));
  ESP_LOGD(TAG, "+CEREG=1 %d", modem.waitResponse());

  modem.sendAT(GF("+COPS=?"));
  ESP_LOGD(TAG, "+COPS=? %d", modem.waitResponse(20000L, GF("+COPS:")));

  modem.sendAT(GF("+COPS=1,2,\"44010\""));
  ESP_LOGD(TAG, "+COPS=1,2,\"44010\" %d", modem.waitResponse());

  // checking status
  modem.sendAT(GF("+COPS?"));
  ESP_LOGD(TAG, "+COPS? %d", modem.waitResponse());

  // APN
  modem.sendAT(GF("+CGAUTH=1,2,\""), gprsPass, GF("\",\""), gprsUser, '"');
  modem.waitResponse();
  // checking status
  modem.sendAT(GF("+CGAUTH?"));
  ESP_LOGD(TAG, "+CGAUTH? %d", modem.waitResponse(GF("+CGAUTH:")));

  ////////////////////////////////////////
  // TinyGsmClientSIM7600.h
  // copied from bool gprsConnectImpl
  modem.sendAT(GF("+CGDCONT=1,\"IP\",\""), apn, '"', ",\"0.0.0.0\",0,0");
  modem.waitResponse();

  modem.sendAT(GF("+CIPMODE=0"));
  modem.waitResponse();

  modem.sendAT(GF("+CIPSENDMODE=0"));
  modem.waitResponse();

  modem.sendAT(GF("+CIPCCFG=10,0,0,0,1,0,75000"));
  if (modem.waitResponse() != 1) {
    return;
  }

  modem.sendAT(GF("+CIPTIMEOUT="), 75000, ',', 15000, ',', 15000);
  modem.waitResponse();

  modem.sendAT(GF("+NETOPEN"));
  if (modem.waitResponse(75000L, GF(GSM_NL "+NETOPEN: 0")) != 1) {
    return;
  }
  // TinyGsmClientSIM7600.h
  // bool gprsConnectImpl
  ////////////////////////////////////////

  ESP_LOGD(TAG, "GPRS status: %s",
           modem.isGprsConnected() ? "接続中 ..." : "接続していない");

  IPAddress local = modem.localIP();
  ESP_LOGD(TAG, "Local IP: %s", local.toString().c_str());

  ESP_LOGD(TAG, "Signal Status: %d", modem.getRegistrationStatus());
}

void loop() {
  ESP_LOGD(TAG, "...");
  delay(1000);
}
```

Here, I am gonna explain one thing I noticed.

https://www.simcom.com/product/SIM7600X-PCIE.html
First, download documents from here.

- SIM7500_SIM7600 Series_AT Command Manual_V3.00
- SIM7500_SIM7600_Series_HTTP(S)_Application Note_V3.00
- SIM7600_Series_PCIE_Hardware_Design_V1.03

And find this **AT+CGAUTH** page.  

[image]

It shows the parameter order is **password** then **user ID**.  However, https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmClientSIM7600.h#L319 at this line, it is the other way around.

Maybe a lot of APNs uses the same USER / PASS values, so it's not a problem??
After I noticed the difference, I switched my USER / PASS parameter position, and I think I remember it started to working.

Hope this helps.

kyotee89, 2024-03-05:
@github0013 thanks for response mate ill take a look tonight. Appreciate your time : )