Xinyuan-LilyGO/LilyGo-T-PCIE #29: sim7600SA -Update firmware - Issue sim7600E

McStick0, 2023-01-29:
Hello!
About a year ago I ordered a package ( but due to customs reasons, I did not receive it for a very long time ) which contained ESP32-Wrover-B Pcie+SimCom 7600E Pcie.

I connected the Pcie as in the tutorial ( [https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/doc/sim7600SA_update_firmware.md](url) ) but on the computer no SimCom showed up in my device list.
[image]

I tested the board for WifiScan, when connected to usb-c the LED on the simcom also lights up.
Any advice?

lewisxhe, 2023-02-04:
You are connected incorrectly. DM (D -) and DP (D+) should be directly connected to the D+and D - of the computer instead of USB TO TTL

McStick0, 2023-02-05:
@lewisxhe 
I understood that a regular USB cable and connect like?:
Red Vcc -- VBus
White Data(+) -- DM
Green Data(-) -- DP
Black GND -- GND

while writing this, I did as I understood - I still see only COM3 and COM4 in the system (some small success )

McStick0, 2023-02-05:
I tried to connect according to this diagram. 
https://docs.espressif.com/projects/esp-at/en/latest/esp32/Get_Started/Hardware_connection.html

Apparently the update went, when I downloaded the ATComandTester for SimCom Modem, and I can't connect to one, and the other is still busy
[image]

McStick0, 2023-02-05:
@lewisxhe well, stuck again. Can't click browse.
But step forward, installed driver
[image]

lewisxhe, 2023-02-06:
Try the latest V1.67 tool  
https://1drv.ms/u/s!AmbpOqVezk5dsULM1jyhu4S--GpX?e=uyhSYc

McStick0, 2023-02-06:
@lewisxhe nope. 
The same thing

lewisxhe, 2023-02-07:
Please open the compatibility mode test. I have tested that all three computers on my side can work normally. If you can, please replace the computer for test

closed 2023-02-11 by lewisxhe