Xinyuan-LilyGO/LilyGo-T-PCIE #30: how to set Default voltage for AXP192!

khoa21214475, 2023-07-20:
[AXP192 Datasheet v1.13_cn.pdf](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/files/12102488/AXP192.Datasheet.v1.13_cn.pdf)
In section 9.6 of this datasheet (9.6 默认电压/启动时序的设置(Default Voltage/Timing Setting)),  how can I set Default voltage, this doccument say "For this part, please refer to the "Default Configuration Instructions" document", so I don't know where can I find it.

Please help me! I want to set DCDC1 to 3.3V and DCDC3 to 3.3V when AXP192 startup!

lewisxhe, 2023-07-21:
The default configuration is chip customization function and cannot be rewritten

khoa21214475, 2023-09-02:
This means that only the manufacturer can configure the chip for the first time, right, I bought 100 chips but can't use it.

lewisxhe, 2023-09-02:
Yes . This is the factory default of the chip and cannot be changed.

khoa21214475, 2023-09-04:
What about chip AXP2101? DCDC1 when first start up is definitely 3.3VDC according to the AXP2101 datasheet, right?

Thank you very much!
[AXP2101.pdf](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/files/12508469/AXP2101.pdf)

khoa21214475, 2023-09-05:
Please help me!