Xinyuan-LilyGO/LilyGo-T-PCIE #31: Startup sequence in examples wrong

FloBab73, 2023-08-04:
In all the examples the part with the PWR_PIN and IND_PIN is not needed and actually wrong. These pins are described in the documentation of the chip, but they aren't connected to the PCIe interface. The PWR_KEY is activated automatically when powering the module. Especially waiting on the IND_PIN for a status can be fatal, because the status will  never come.

The scematic even states that pin 6 on the pcie interface is the POWERKEY and leads to not connected, which is confirmed, by the documentations of the modules.

[image]

[image]
documentation from sim7600 pcie

fabiopolar, 2023-08-15:
Hi !
I am having several problem no Network Connect, I think it´s related to you issue, I will try some tests without this initialization.

lewisxhe, 2023-09-02:
T-PCIE only supports LilyGo's PCIE module. Others do not support