Xinyuan-LilyGO/LilyGo-T-PCIE #38: Issue with SIM7080 Firmware Update

luimorper, 2024-07-11:
Hi,

I'm experiencing a problem similar to the one described in [[Issue #29](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/issues/29)]. Despite following the recommended procedures, I haven't been able to successfully update the firmware on my SIM7080 module.
Steps I've Taken:

    Driver Installation:
        Installed the drivers for the SIM7080 module.
        Verified the connection with AT commands and received appropriate responses from the module.

    Upgrade Tool:
        Tried different versions of the upgrade tool, including the latest V1.67 as suggested.
        None of these attempts were successful in completing the firmware update.

Symptoms:

    The module is recognized by the computer and responds to AT commands.
    The upgrade tool does not provide any option to browse for the firmware within my computer.

Additional Information:

    I have tried the upgrade process on different computers with the same result.
   

Has anyone encountered similar issues or can offer additional troubleshooting steps? Any help would be greatly appreciated.

Thank you!

[image]

seba-tokbo, 2025-07-21:
Have you been able to resolve the issue? I'm currently experiencing the same problem.

lewisxhe, 2025-07-22:
You can refer to this video https://www.youtube.com/watch?v=lGctqLDGHz4

lewisxhe, 2025-07-23:
Win11 I can confirm it is normal