Xinyuan-LilyGO/LilyGo-T-PCIE #41: Best Micropython Firmware to use with T-PCIE ..

MKoroschetz, 2024-09-10:
Could anyone be so kind to confirm that the [lilygo-micropython](https://github.com/Xinyuan-LilyGO/lilygo-micropython) is the best Micropython firmware for the T-PCIE board? In the description it does not mention this board specifically and it appears to be based on micropython v1.19.
My concern is particularly related to the support of the AXP192 and the extended support for the PCIE-SIM707G which I have for my T-PCIE.
I need to port my Micropython code which is working on the old original MicroPython_LoBo  on the T-CALL SIM800L.
Thank you in advance.

lewisxhe, 2024-09-19:
MicroPython_LoBo firmware can also run on PCI-E motherboards, AXP2101 can be completely ignored, there are also micropython examples https://github.com/lewisxhe/XPowersLib/tree/master/Micropython/examples

MKoroschetz, 2024-10-07:
Thank you for the link of the examples above.
I understand that I can use MicroPython_LoBo. However, I know you also have a specific [LilyGo variant](https://github.com/Xinyuan-LilyGO/lilygo-micropython) which uses a much more recent MicroPython version. Nevertheless, I only see specific config files and references to the T-CALL and T-SIM7000G boards. None for the T-PCIE . 
Thank you in advance for commenting and guiding me in the right direction..

lewisxhe, 2024-10-07:
[lilygo-micropython](https://github.com/Xinyuan-LilyGO/lilygo-micropython)  is currently unmaintained, so I can't help you.

MKoroschetz, 2024-12-31:
Hi there,
I am actively working on a data logger framework based on the T-PCIE with the simcom7070G modem using micropython 1.20 for an agriculture project in Europe.
I would like to update the 7070G modems I have , to the latest firmware, which as per [Simcom ](https://www.simcom.com/technical_files.html) is 1951B16SIM7070 released on 2024-06-17.
Could you please be so kind and update the instructions available for the [LilyGO-T-SIM7000G](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G) specifically for the T-PCIE board.  Hopefully, you also can facilitate the respective firmware file.  Unfortunately, I was not able to find the firmware file on the [Simcom ](https://www.simcom.com/technical_files.html) website.
Thank you in advance for your comments and help.

lewisxhe, 2025-01-02:
@MKoroschetz Please open another issue

lewisxhe, 2025-01-02:
The PN number on the modem, and AT+SIMCOMATI to check the current software version, tell me, I will check if there is a new firmware

MKoroschetz, 2025-01-03:
Hi Lewis,

Thanks for the prompt response. Following the information

'AT+GCAP;+GMI;+GMM;+GMR;+GSN\r\r\n+GCAP:+CGSM\r\n\r\nSIMCOM_Ltd\r\n\r\nSIMCOM_SIM7070\r\n\r\nRevision:1951B11SIM7070\r\n\r\n860371050882459\r\n\r\nOK\r\n'
This indicates Firmware: 1951B11SIM7070

As per attached release document version 1951B16SIM7070 should be available

I am also attaching the latest firmware release notes taken from simcom's website.

The 4 additional PCIE cards have not yet arrived. They shipped out from your warehouse in China i believe 12/29/2024 and are therefor in transit.

Thank you in advance and best regards,

Manfred Koroschetz

lewisxhe, 2025-01-03:
[1951B16SIM7070.rar](https://drive.google.com/file/d/10scjUW4nFGLG06-9R0em6NTgeyjUxlfh/view?usp=sharing)

MKoroschetz, 2025-01-06:
Thank you.
I understand that I could do a FOTA update just by having the file accessible on our server and setting the proper AT+ commands to do it.
Am I correct?
This would facilitate the upgrade process as no physical manipulation and visit to remote locations is required.
Thank you in advance for your comment.

lewisxhe, 2025-01-07:
The provided firmware can only be updated via USB, FOTA needs to be applied to SIMCOM

MKoroschetz, 2025-01-26:
Hi Levis,
I have been able to get the T-PCIE 1.1 and 1.2 to work with stock micropython 1.20. More precisely, I can issue any AT command as per Simcom 7070G AT manual.
However, I have not been able to get a TCP connection to our mosquitto MQTT server working. I am following the application notes, I found. I get the "transparent transport" connection over LTE to our server established. But do not get any trafic from the T-PCIE to the server and vice/versa.
I am pretty sure some minor detail is escaping my attention.
Any chance you have an example code in micropython or for arduino which you know is working?  Or maybe some other suggestion?
Thank you in advance for your help.
Manfred Koroschetz

MKoroschetz, 2025-01-26:
Hi Levis,
on a different note. I have put one of my 7070G modules to a higher baudrate (I believe 230400) using the AT+IPR=230400 command.
And now I am unable to issue any commands to the modem anymore. All intents to revert to 115200 have failed so far.
Is there a known way to reset the 7070G to factory settings which I believe set the modem to Auto baud?
Thanks
Manfred Koroschetz

lewisxhe, 2025-02-06:
> Hi Levis, I have been able to get the T-PCIE 1.1 and 1.2 to work with stock micropython 1.20. More precisely, I can issue any AT command as per Simcom 7070G AT manual. However, I have not been able [...]

As far as I know, the built-in MQTT is very difficult to use. I have tried several times and failed. Please check the manual. I can't help with this problem. Or you can refer to the SIM7080 one. The two modules and commands are very similar. https://github.com/Xinyuan-LilyGO/LilyGo-T-SIM7080G

lewisxhe, 2025-02-06:
> Hi Levis, on a different note. I have put one of my 7070G modules to a higher baudrate (I believe 230400) using the AT+IPR=230400 command. And now I am unable to issue any commands to the modem [...]

The recovery method is to connect the USB of the MODEM to the computer and reset the baud rate through the USB AT port.