Xinyuan-LilyGO/LilyGo-T-PCIE #42: Battery polarity

vpanadeiro, 2024-09-18:
Battery polarity
How can I know which is the negative and positive of the battery, does it have a mark on the PCB for the positive or the negative? I have doubts because the cable that comes with the board seems to be the opposite of what usually comes with the battery.

I don't know if you have any documentation on the general pinout of the board.

lewisxhe, 2024-09-19:
Are the included cables opposite to the actual PCB?
[image]

vpanadeiro, 2024-09-20:
Thank you for your quick response, very clarifying. What I meant is that standard batteries usually come with the polarity reversed, like this one.
[image]

 
Is it possible to measure the battery voltage through an ADC?

lewisxhe, 2024-09-21:
https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/examples/PMUExample/PMUExample.ino