Xinyuan-LilyGO/LilyGo-T-PCIE #43: I want to make a phone number ring with a T-SIM7670G

23tom2323, 2024-11-28:
Hello,
I want to make a phone number ring with a T-SIM7670G.

I tested these examples for the first time.
LilyGo-T-PCIE/examples/A7670/TinyGSM_Net_GNSS/TinyGSM_Net_GNSS.ino
I have to choose a GNSS mode for this, but which one?
I only need the phone number to ring. No GPS or other functions.
What mode do I need? the following are available:

gnss_mode:

0 : GLONASS

1 : BEIDOU

2 : GALILEO

3 : QZSS

dpo_mode :

0 disable

1 enable

Thanks for the help in advance

23tom2323, 2024-11-28:
I think LilyGo-T-PCIE/examples/A7670/TinyGSM_Net_GNSS/TinyGSM_Net_GNSS.ino
is not for a call...

Does anyone have an example of a call / alarm call?

I thought it would work with ATD+ +4921649500000

I'm also not sure if I really need a microphone and a speaker. The Sim800 should have one... but it works great without it.

lewisxhe, 2024-11-28:
All versions of SIM7670G do not support voice

lewisxhe, 2024-11-28:
SIM7670G is here https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX, not in this repository