Xinyuan-LilyGO/LilyGo-T-PCIE #44: High power consumption in Deep Sleep

Maestro-NL, 2025-03-20:
--

Testing with the Arduino ESP32 Deep Sleep example the board is using ~1.8mA during deep sleep. 
While testing there was no T-PCIE SIM module placed, just the bare T-PCIE board connected to a 18650 cell (@4.15V)

I was hoping to get lower power consumption in deep sleep (below 1mA, [as shown here](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/examples/SIM7600/sleep/modem%20sleep.jpg))
The board is a V1.2 (23-5-14) with AXP2101. Any advice on getting power consumption down? 

[image]

lewisxhe, 2025-03-22:
Provide your test code

Maestro-NL, 2025-03-22:
I have used the default Arduino ESP32 example, with only a modification for turning LED on GPIO12 off:

[code]
#define LED_PIN 12
#define uS_TO_S_FACTOR 1000000ULL /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  15          /* Time ESP32 will go to sleep (in seconds) */

RTC_DATA_ATTR int bootCount = 0;

void setup() {
  Serial.begin(115200);
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, LOW); // Turning on LED 
  delay(3000);  //Take some time to open up the Serial Monitor

  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
  print_wakeup_reason();

  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) + " Seconds");

  Serial.println("Going to sleep now");
  digitalWrite(LED_PIN, HIGH); //Turning off LED
  Serial.flush();
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop() {
  //This is not going to be called
}
[/code]

The A7608E-H modem was temporarily removed to rule  it out as power consumer in this test.

lewisxhe, 2025-03-24:
I'll upload a sample, you can try it, I don't have a board yet, so I can't test it

Maestro-NL, 2025-03-24:
Thank you, now we're talking micro-amps!
Power consumption is considerably lower when loading this code, about 400µA.

I was already hoping the AXP2101 was causing the drain and it could be lowered thru software parameters.
Although i wasn't confident with the chip and parameters (yet) to fiddle it out myself. Now i am :)

For what it's worth: 
Disabling the MODEM DCDC BOOSTER brings consumption down from 1800 to 1000µA
Also disabling all power channels (Fixed and LDO) brings consumption down from 1000 to 500µA
Your DeepSleep.ino example as given** brings consumption down to 400µA. (Modem attached)

[image]

** Line 33 needed a change to compile succesfully: 
```
- if (!power.init(Wire, AXP2101_SLAVE_ADDRESS, i2c_sda, i2c_scl)) {
+ if (!power.begin(Wire, AXP2101_SLAVE_ADDRESS, i2c_sda, i2c_scl)) {
```

lewisxhe, 2025-03-25:
Yes, I have changed it. I just got the board to test it today.
[image]

closed 2025-04-05 by Maestro-NL